robotology/gazebo-yarp-plugins #694: Failure when trying to compile gazebo-yarp-plugins

martinaxgloria, 2025-02-17:
Today I was trying to compile `gazebo-yarp-plugins v4.11.2` on WSL2 with `Ubuntu 22.04`, but it failed with the following error:

```sh
Consolidate compiler generated dependencies of target gazebo_yarp_rpc_linkattacher
Consolidate compiler generated dependencies of target gazebo_yarp_rpc_worldinterface
Consolidate compiler generated dependencies of target gazebo_yarp_showmodelcom
Consolidate compiler generated dependencies of target gazebo_yarp_modelposepublisher
Consolidate compiler generated dependencies of target gazebo_yarp_rpc_clock
Consolidate compiler generated dependencies of target gazebo_yarp_externalwrench
Consolidate compiler generated dependencies of target gazebo_yarp_MultiCameraPlugin
[  4%] Built target gazebo_yarp_rpc_worldinterface
[  6%] Built target gazebo_yarp_rpc_linkattacher
[  8%] Built target gazebo_yarp_rpc_clock
[  9%] Building CXX object libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o
[ 11%] Built target gazebo_yarp_modelposepublisher
Consolidate compiler generated dependencies of target gazebo_yarp_clock
[ 13%] Built target gazebo_yarp_showmodelcom
[ 14%] Built target gazebo_yarp_MultiCameraPlugin
[ 15%] Building CXX object plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/src/ApplyExternalWrench.cc.o
[ 16%] Building CXX object plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/src/ExternalWrench.cc.o
[ 19%] Built target gazebo_yarp_clock
In file included from /usr/include/gazebo-11/gazebo/common/common.hh:21,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
                 from /home/mgloria/iit/robotology-superbuild/src/GazeboYARPPlugins/plugins/externalwrench/include/gazebo/ExternalWrench.hh:18,
                 from /home/mgloria/iit/robotology-superbuild/src/GazeboYARPPlugins/plugins/externalwrench/src/ExternalWrench.cc:8:
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:52:44: error: expected unqualified-id before ‘)’ token
   52 |       public: MovingWindowFilterPrivate<T>();
      |                                            ^
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:73:5: error: no declaration matches ‘gazebo::common::MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate()’
   73 |     MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate() :
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:73:5: note: no functions named ‘gazebo::common::MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate()’
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:49:11: note: ‘class gazebo::common::MovingWindowFilterPrivate<T>’ defined here
   49 |     class MovingWindowFilterPrivate
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:88:37: error: expected unqualified-id before ‘)’ token
   88 |       public: MovingWindowFilter<T>();
      |                                     ^
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:124:5: error: no declaration matches ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter()’
  124 |     MovingWindowFilter<T>::MovingWindowFilter()
      |     ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:115:27: note: candidate is: ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter(gazebo::common::MovingWindowFilterPrivate<T>&)’
  115 |       protected: explicit MovingWindowFilter<T>(
      |                           ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:85:11: note: ‘class gazebo::common::MovingWindowFilter<T>’ defined here
   85 |     class MovingWindowFilter
      |           ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/gazebo-11/gazebo/common/common.hh:21,
                 from /home/mgloria/iit/robotology-superbuild/src/GazeboYARPPlugins/libraries/singleton/src/ConfHelpers.cc:13:
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:52:44: error: expected unqualified-id before ‘)’ token
   52 |       public: MovingWindowFilterPrivate<T>();
      |                                            ^
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:73:5: error: no declaration matches ‘gazebo::common::MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate()’
   73 |     MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate() :
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:73:5: note: no functions named ‘gazebo::common::MovingWindowFilterPrivate<T>::MovingWindowFilterPrivate()’
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:49:11: note: ‘class gazebo::common::MovingWindowFilterPrivate<T>’ defined here
   49 |     class MovingWindowFilterPrivate
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:88:37: error: expected unqualified-id before ‘)’ token
   88 |       public: MovingWindowFilter<T>();
      |                                     ^
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:124:5: error: no declaration matches ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter()’
  124 |     MovingWindowFilter<T>::MovingWindowFilter()
      |     ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:115:27: note: candidate is: ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter(gazebo::common::MovingWindowFilterPrivate<T>&)’
  115 |       protected: explicit MovingWindowFilter<T>(
      |                           ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:85:11: note: ‘class gazebo::common::MovingWindowFilter<T>’ defined here
[... 18 lines cut ...]
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:88:37: error: expected unqualified-id before ‘)’ token
   88 |       public: MovingWindowFilter<T>();
      |                                     ^
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:124:5: error: no declaration matches ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter()’
  124 |     MovingWindowFilter<T>::MovingWindowFilter()
      |     ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:115:27: note: candidate is: ‘gazebo::common::MovingWindowFilter<T>::MovingWindowFilter(gazebo::common::MovingWindowFilterPrivate<T>&)’
  115 |       protected: explicit MovingWindowFilter<T>(
      |                           ^~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-11/gazebo/common/MovingWindowFilter.hh:85:11: note: ‘class gazebo::common::MovingWindowFilter<T>’ defined here
   85 |     class MovingWindowFilter
      |           ^~~~~~~~~~~~~~~~~~
make[2]: *** [libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/build.make:90: libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:658: libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/build.make:76: plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/src/ApplyExternalWrench.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/build.make:90: plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/src/ExternalWrench.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:840: plugins/externalwrench/CMakeFiles/gazebo_yarp_externalwrench.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

Here the `apt list --installed`:

[apt-list.txt](https://github.com/user-attachments/files/18826837/apt-list.txt)

traversaro, 2025-02-17:
Which YARP version are you using? Ubuntu 22.04 with apt dependencies is tested in the CI of the superbuild, and the Gazebo version seems the default one, so I wonder if there is anything else "strange" in your system.

martinaxgloria, 2025-02-18:
Actually, by recompiling the robotology-superbuild from a fresh clone I was able to compile everything without any error, so probably something strange happened to my environment. Thanks @traversaro, closing then

closed 2025-02-18 by martinaxgloria